mhamidjamil/TTGO_TCall #1: Unable to read all messages correctly

hamidjamil0420, 2023-07-20:
when we send all command via serial terminal we get ### incorrect response, like last two pulls causes this issues so try to revert them or create new branch from main (_develop_) and implement same changing of _staging_ but monitor the response after each changes and then merge that branch (_develop_) to _staging_ and _main_ if it works fine, and one more thing don't forgot to mention the issue.

mhamidjamil, 2023-07-21:
its fixed in over latest code version. In latest code when module receive message it execute it and delete it (if that message is executed successfully) so module don't have to store useless messages now, and it receives any message which is not understandable then it will send that message to @mhamidjamil and ask what to do with it, its on @mhamidjamil that he want module to keep the message or not (in 90% cases he ask module to delete it) so by doing this i and message forwarding

mhamidjamil, 2023-07-21:
but remember this last feature is not in any branch it is still in development, but this problem #2  is solved in leatest version

closed 2023-07-21 by mhamidjamil