mhamidjamil/TTGO_TCall #95: Messaging part is not working!

mhamidjamil, 2023-09-18:
**Describe the bug**
When module power up or it receive message it is not forwarding it or no message in a day, it is not possible that the module didn't receive any message from company. 

**To Reproduce**
Steps to reproduce the behavior:
1. Call it or reboot it you ahould get a message but not

**Expected behavior**
When module reboot or recive call it should send a message and yes one more thing also use these WhatsApp messages feature.

wasey-rao, 2023-09-19:
I have checked, it happened because sim nether have message package nor balance.

mhamidjamil, 2023-09-19:
I think then #61 part is useless until it can sense that module ran out of balance or package.
create new feature for this issue. so, in future i should not send message when the package date over..

closed 2023-09-19 by mhamidjamil