mhamidjamil/TTGO_TCall #96: Auto update package expiry date to save balance and to make sure code flow work as expected.

wasey-rao, 2023-09-19:
**Feature request related to a problem #95**
Module's Sim package was expired on say 14 and today it's 19, I am wondering why module is not sending or forwarding any messages, after investigation I came to know that it ran out off balance so #61 is useless until module automatically stop message sending feature.

**Solution**
Create two new fields on thingSpeak.

1. First field maintain the date when the package will expires. (use the attached screenshot to update this field)
2. Second field will be one who will change the bool variable => either message part will work or not, mean sim have subscribed to package or not.

**Alternatives you can considered**
Use SPIFFS memory store the package expiry date in txt file, so when module power up and if it don't connect with WIFI to fetch these two package information related fields then it should use SPIFFS date and wait until any message arrive then update its date and time and after that it should decide either it should use messaging part or not.

**Additional context**
Regarding thingSpeak and SPIFFS package expire date update use attached screenshot message.
module should use this message carefully, to update its parameters of thingSpeak and SPIFFS. 
[image]
as I use this package so you can use highlighted part of message to update the expiry date but the better practice is to use underline part of message so it might work as a generic solution.

wasey-rao, 2023-09-19:
This could be the last feature update for version 5 after this only bug fixes will be supported for 1 week till first of October.

wasey-rao, 2023-09-19:
@mhamidjamil please ignore the second part of solution you don't need to create a bool type variable on thingSpeak you just fetch package expire date and compare it with local time and then take decision on local system, no need to fetch that part from thingSpeak every time.

mhamidjamil, 2023-09-22:
https://github.com/mhamidjamil/TTGO_TCall/pull/102/

closed 2023-09-25 by mhamidjamil