mhamidjamil/TTGO_TCall #112: Unable to send whatsapp messages.

wasey-rao, 2023-10-03:


wasey-rao, 2023-10-03:
Unable to whatsapp functionality via BLE and some time with BLE, i get -1 http error code, 

[image]

 is the screen shot

mhamidjamil, 2023-10-03:
I checked [image]this issue; it is almost working fine if we just use serial terminal but failed when we use BLE.

mhamidjamil, 2023-10-03:
I think BLE cause some kind of disturbance in WIFI connectivity, so the WIFI part doesn't work as expected, still gathering more information.
my conclusion is:
when we connect with BLE WIFI don't work as expected that's why the http request failed,
[image]

mhamidjamil, 2023-10-03:
My conclusion is: enqueue Wi-Fi related jobs and execute when BLE device disconnect and also suggest that if BLE connected device is not sending any data for a specific duration then it should be disconnected.

closed 2023-11-11 by wasey-rao