mhamidjamil/TTGO_TCall #152: Auto send message to unknow caller.

mhamidjamil, 2024-01-06:


mhamidjamil, 2024-01-06:
When module receives call from unknow number it should reject the call immediately and send message to that caller too to contact the owner number, currently it just sends message to owner and don't cut the call, it would be a best option to manage this from orange pi, what do you think @wasey-rao?

wasey-rao, 2024-01-06:
Regarding you question No, I don't think we should manage this from orange pi what if only ttgo is in working state and orange pi is disconnecteed due to any reason? I think we should implement this functionality in ttgo module only just ping orange pi and send data to save records not to manage this functionality and I think you should first make sure #136 this issue is close as a lot of issues can be reported due to this issue, so please do complete it as soon as possible. please do let me know which calls will be white/black listed for this issue @mhamidjamil

mhamidjamil, 2024-01-06:
I follow these rules:
1. Don't send message to company calls (length of caller number less then 10)
2. Don't send messages to PTCL number stat from 042 or any.

- It only sends message to mobile number with length greater then 10 and start from `+923` or `03`
- I filter out the annoying number like if some one call more then 3 times a day then I mark that as a black listed number and will not send message to that number.

these are the points I'll follow and hope fully I close this issue on this weekend, do you have any other suggestions @wasey-rao  and you can mark these points a task list.

closed 2024-01-08 by wasey-rao