mhamidjamil/TTGO_TCall #163: Module is not deleting company messages

wasey-rao, 2024-02-15:


wasey-rao, 2024-02-15:
Module is not deleting company message by it self after #162 here is the attached image:

[image]

I think it is because you have moved company numbers to the SPIFFS and you might not have added these as a company numbers am i right? @mhamidjamil

mhamidjamil, 2024-02-15:
Absolutely we were deleting messages based if Jazz is present in the sender number now we are not doing this we are mapping complete numbers, there are two solutions what do you want to me to go with:

1. As scheduled task #159 it might take time to implement this as currently our priority is to implement [NTFY](https://github.com/mhamidjamil/orangePi/issues/105) I will complete by this weekend and start working on this issue.
2. I should add these numbers to SPIFFS no code update required we just need to add using serial terminal.

waiting for you response option two will be implemented in a minute. @wasey-rao

wasey-rao, 2024-02-15:
Please go for option 2 for now (as it is annoying to get these company messages continuously) but make sure to complete #159 too. @mhamidjamil

mhamidjamil, 2024-02-15:
Sure, can you please send me company number which you have pointed after this update and please merge this pr too as early as possible as i have to make other changes too.

wasey-rao, 2024-02-15:
`Jazz 4G` and `Jazz Lahore`

mhamidjamil, 2024-02-15:
@wasey-rao i have added these number to company message list but not sure if module can process them or not.
[image]

wasey-rao, 2024-02-15:
Ok, thanks hope it work let see.

closed 2024-03-15 by wasey-rao