mhamidjamil/TTGO_TCall #164: Make an app to send messages to other number using BLE.

mhamidjamil, 2024-02-19:


mhamidjamil, 2024-02-19:
@wasey-rao Can you please make an app which connect with this module with BLE tech and it has two input fields in first field user enter receiver number and in second user enter message and send it, that message should be convert in a string like:
`smsTo {message_body} [number_number] `

mhamidjamil, 2024-08-20:
@SaffiUllah1999 can you please continue this i'll provide you API in a day, please also let me know when you will implement this feature so i create a millstone for mobile app work too.

mhamidjamil, 2024-08-20:
Hi @SaffiUllah1999,
Would it be possible to set a milestone for October 1st? This gives us four weeks to ensure the app is ready, but I'm open to any alternative dates you might suggest.

SaffiUllah1999, 2024-08-20:
Hi @mhamidjamil 
Please update the title of the ticket and also provide the API to send message.

mhamidjamil, 2024-08-20:
@SaffiUllah1999 I have updated title as per discussion I'm creating API for messaging I'll let you know when it is ready.

SaffiUllah1999, 2024-09-25:
@mhamidjamil  Please my Postman Account saffiullah1911@gmail.com

mhamidjamil, 2024-09-25:
@SaffiUllah1999 I have sent you an invite please accept it and let me know if you need any assistance on APIs

SaffiUllah1999, 2024-09-25:
@mhamidjamil  I have accepted the invite and app will be ready by tomorrow

mhamidjamil, 2024-09-25:
@SaffiUllah1999 Take your time, can you please make sure these points:
- [ ]  User can change receiver number.
- [ ]  User can write custom message.
- [ ]  (Optional) Adds send by with the message, say if user write message: "This is test message" app should have any global variable which will update this message to "Message from: [Username]: This is test message"

SaffiUllah1999, 2024-09-26:
@mhamidjamil As you have mention in point 3 ; use global variable to update specific case only?

mhamidjamil, 2024-09-26:
@SaffiUllah1999 not just in specific case, I will be like an addition to a message body, like we have signature in email, we set it somewhere else and when we send email that signature also added to the email, I'm talkin about that part for our message body

SaffiUllah1999, 2024-09-26:
@mhamidjamil Task is done, now i am working on some cosmetic improvements

SaffiUllah1999, 2024-09-26:
@mhamidjamil  I have build the apk here is the link: https://expo.dev/artifacts/eas/3ZmTxsfD6akhJiNpsVRCS8.apk

SaffiUllah1999, 2024-09-26:
@mhamidjamil I have tested the apk, it is throwing error, please have a look

mhamidjamil, 2024-09-29:
@SaffiUllah1999 Can you please check now, I have updated the hardware now it should be working fine as expected.

mhamidjamil, 2024-09-29:
@SaffiUllah1999  Please make sure you create versions and a milestone in order to complete the requirements.

SaffiUllah1999, 2024-09-29:
@mhamidjamil Are you handling the error message in case of 500 (internal server error) & 400 (bad request) ?

mhamidjamil, 2024-09-29:
@SaffiUllah1999 yes, please let me know if you face any error or unexpected behaviour from backend.